SKAARHOJ/skaarhoj-updater-releases #4: Issue with GPIO module

alexgmoonFTS, 2020-01-17:
Background Issue: 
      
```
Error: Object has been destroyed
    at BrowserWindow.setProgressBar (<anonymous>)
    at Bossa.<anonymous> (/Applications/SKAARHOJUpdaterBeta.app/Contents/Resources/app.asar/background.js:37:322050)
    at Bossa.emit (events.js:205:15)
```

---

SKAARHOJUpdaterBeta 0.9.19
Electron 6.1.4
darwin 18.7.0
Locale: en-US

s00500, 2020-01-17:
Hi Alex, thanks for reporting this issue,
Did you close the window at any point ? or can you post the logfile ? It would be interesting how this problem can be reproduced reliably?

Are you on windows ? or some other OS ?

Greetings

s00500, 2020-01-17:
Duplicate of #3 in different environment

alexgmoonFTS, 2020-01-17:
I closed the window, and restarted the software and it worked correctly.

I am on OSX 10.14.6
*Alex G. Moon, CTS-D*

Director of Engineering and Service, Force Technology Solutions

O: 317.586.2844 | M: *317.919.5173* | www.forcetechsolutions.com |
6097 W 400 N STE C. GREENFIELD, IN 46140-6201

[image: https://s3.amazonaws.com/images.wisestamp.com/icons_32/facebook.png]
<http://www.facebook.com/forcetechsolutions/> [image:
https://s3.amazonaws.com/images.wisestamp.com/icons_32/linkedin.png]
<http://www.linkedin.com/in/alex-moon-2118592b>

CONFIDENTIALITY NOTICE:
This email message sent by AGMoon@force.tech or AGMoon
@forcetechsolutions.com and any accompanying data or files
are confidential and may contain privileged information intended only for
the named recipient(s). If you are not the intended recipient(s), you are
hereby notified that the dissemination, distribution, and or copying of
this message is strictly prohibited. If you receive this message in error,
or are not the named recipient(s), please notify the sender at the email
address above, delete this email from your computer, and destroy any copies
in any form immediately. Receipt by anyone other than the named
recipient(s) is not a waiver of any attorney-client, work product, or other
applicable privilege.

s00500, 2020-01-24:
v0.9.20 should fix the issue completely

s00500, 2020-02-10:
@alexgmoonFTS Can you confirm this is fixed? Otherwise just reopen this issue, Thanks

closed 2020-02-10 by s00500