SKAARHOJ/skaarhoj-updater-releases #6: Check For Updates Error

unnamedculprit, 2020-01-23:
While checking for updates for Rack Fusion Live, popup error keeps appearing:

Background Issue: 
      
```
Error: Object has been destroyed
    at BrowserWindow.setProgressBar (<anonymous>)
    at Bossa.<anonymous> (/Applications/SKAARHOJUpdaterBeta.app/Contents/Resources/app.asar/background.js:37:322050)
    at Bossa.emit (events.js:205:15)
```

---

SKAARHOJUpdaterBeta 0.9.19
Electron 6.1.4
darwin 18.7.0
Locale: en-GB

unnamedculprit, 2020-01-23:
[image]

s00500, 2020-01-24:
Hey, thanks for reporting, we are looking into it already
duplicate of #4

s00500, 2020-01-24:
v0.9.20 should fix this issue
Knew Tom Scotts youtube for a long while, now I know yours 👍
Greetings

s00500, 2020-02-10:
@MattGrayYes Can you confirm this issue is fixed ? Otherwise just reopen here

closed 2020-02-10 by s00500

MattGrayYes, 2020-02-10:
Yep that’s fixed it for me!
Thanks.