SKAARHOJ/skaarhoj-updater-releases #10: Correct message when internet is unavailable

pjnr1, 2020-08-10:
When launching the firmware update, without internet connection I get the following error message:
[image]

Same goes if checking for updates later on (they program starts normally after clicking okay.)

Seems that the server name for auto-updates isn't resolved. Should have a better error name since it's probably not that uncommon.

s00500, 2020-08-24:
Pretty strange as I have patched this issue in a previous release, what version are you using ?

pjnr1, 2020-08-24:
[image]

s00500, 2020-08-24:
Thanks, I have fixed the issue, it will be included in the next release

closed 2020-08-24 by s00500

pjnr1, 2020-08-25:
Fantastic, thanks :)