SKAARHOJ/skaarhoj-updater-releases #11: PTZ FLY Setup

productionspec22, 2020-09-06:
Background Issue: 
      
```
TypeError: Cannot read property 'then' of undefined
    at EventEmitter.<anonymous> (/Applications/SKAARHOJUpdater.app/Contents/Resources/app.asar/background.js:16:330291)
    at EventEmitter.emit (events.js:200:13)
    at WebContents.<anonymous> (/Applications/SKAARHOJUpdater.app/Contents/Resources/electron.asar/browser/api/web-contents.js:335:21)
    at WebContents.emit (events.js:200:13)
```

---

SKAARHOJUpdater 1.0.1
Electron 6.1.10
darwin 19.6.0
Locale: en-US

s00500, 2020-09-07:
Hi @productionspec22 ,
Same for this one,
Could you explain a little what you did to trigger this issue in more detail ?
Greetings,
Lukas

s00500, 2020-09-08:
like #12 this should be fixed in v1.0.2

closed 2020-09-08 by s00500