SKAARHOJ/skaarhoj-updater-releases #12: PTZ flY 2

productionspec22, 2020-09-06:
Background Issue: 
      
```
TypeError: Cannot read property 'write' of null
    at SerialPort.<anonymous> (/Applications/SKAARHOJUpdater.app/Contents/Resources/app.asar/background.js:16:311248)
    at SerialPort._error (/Applications/SKAARHOJUpdater.app/Contents/Resources/app.asar/node_modules/@serialport/stream/stream.js:197:14)
    at /Applications/SKAARHOJUpdater.app/Contents/Resources/app.asar/node_modules/@serialport/stream/stream.js:241:12
```

---

SKAARHOJUpdater 1.0.1
Electron 6.1.10
darwin 19.6.0
Locale: en-US

s00500, 2020-09-07:
Hi @productionspec22 ,
Could you explain a little what you did to trigger this issue in detail ?
Greetings,
Lukas

s00500, 2020-09-07:
Ok nevermind, you have likely stumbled upon those issues when something in the controller has gone wrong, try powercycling or if that does not help a full erase. Both issues have been fixed in v1.0.3 and will now generate proper errors, thanks for your report =D

closed 2020-09-07 by s00500