SKAARHOJ/skaarhoj-updater-releases #13: PTZ Fly can't connect

kieranconnellan, 2020-10-14:
Background Issue: 
      
```
Error: net::ERR_CONNECTION_RESET
```

---

SKAARHOJUpdater 1.0.1
Electron 6.1.10
win32 10.0.17763
Locale: en-US

s00500, 2020-10-15:
Hello,
Thanks for reporting the issue,
Please try to upgrade to the newest version of the Updater (v1.0.4)
If the issue remains please try to attach the full log file to this report, thanks 😊 

Lukas

kieranconnellan, 2020-10-15:
The log was sent from the most recent version. At least that is the version
I have installed and that I then opened and created the logs using.

s00500, 2020-10-16:
The automatically generated issue report notes otherwise though: `SKAARHOJUpdater 1.0.1`

If you are getting the same issue with 1.0.4 please attach the log file from this version (In the menu Options-> Show Log)
Best Regards
Lukas

s00500, 2020-10-27:
@kieranconnellan any news on this ? Please reopen this issue if so

closed 2020-10-27 by s00500

kieranconnellan, 2020-10-27:
Looks like it was a firewall issue. Asked Heather to let me know if there
are any exclusions we can add to our firewall to allow it to work through.

Cheers

s00500, 2020-10-27:
Alright, thanks for the update

Greetings,
Lukas