SKAARHOJ/skaarhoj-updater-releases #19: Error on windows 11

McHauge, 2021-11-29:
Background Issue: 
      
```
Error: dns service error: service not running
    at new Browser (C:\Users\mc-ha\AppData\Local\Programs\SKAARHOJUpdater\resources\app.asar\node_modules\mdns\lib\browser.js:88:10)
    at Object.create [as createBrowser] (C:\Users\mc-ha\AppData\Local\Programs\SKAARHOJUpdater\resources\app.asar\node_modules\mdns\lib\browser.js:116:10)
    at Timeout.ce [as _onTimeout] (C:\Users\mc-ha\AppData\Local\Programs\SKAARHOJUpdater\resources\app.asar\background.js:2:6932803)
    at listOnTimeout (internal/timers.js:531:17)
    at processTimers (internal/timers.js:475:7)
```

---

SKAARHOJUpdater 1.0.11-pre2
Electron 6.1.10
win32 10.0.22000
Locale: en-GB

McHauge, 2021-11-29:
Downloaded this:
https://support.apple.com/kb/DL999?locale=en_US

and it began working, but should it not be linked inside the windows version, that you will need it, or be installed as part of the installation?

s00500, 2021-11-29:
Hi @McHauge,
Yes, great point, I will add that

closed 2021-11-29 by McHauge