SKAARHOJ/skaarhoj-updater-releases #20: Error

simonegle, 2022-01-25:
Background Issue: 
      
```
Error: dns service error: service not running
    at new Browser (C:\Users\Simon Egle\AppData\Local\Programs\SKAARHOJUpdater\resources\app.asar\node_modules\mdns\lib\browser.js:88:10)
    at Object.create [as createBrowser] (C:\Users\Simon Egle\AppData\Local\Programs\SKAARHOJUpdater\resources\app.asar\node_modules\mdns\lib\browser.js:116:10)
    at Timeout.ce [as _onTimeout] (C:\Users\Simon Egle\AppData\Local\Programs\SKAARHOJUpdater\resources\app.asar\background.js:2:6933573)
    at listOnTimeout (internal/timers.js:531:17)
    at processTimers (internal/timers.js:475:7)
```

---

SKAARHOJUpdater 1.0.11
Electron 6.1.10
win32 10.0.22000
Locale: de

s00500, 2022-01-25:
Hi Simon, 
This is currently happening when your PC is missing some package to facilitate network discovery. You can eiterh savely ignore it or go and install this package from here:
https://support.apple.com/kb/dl999?locale=en_US

For now I have added a note to the main page of this repository. Later I will add a new error message explaining the issue better, its on my list

Hope that helps,
Lukas

closed 2022-01-25 by s00500

gubrist, 2022-01-29:
You can eithrh savely ignore it 
---> no, I cannot connect my Skaarhoj PTZ Fly any more (Windows 10)
---> They same problems happens on the brand new PC of my wife (Windows 11)

Install this package from here: https://support.apple.com/kb/dl999?locale=en_US
---> I am puzzled to have to install a software from Apple " Download Bonjour Print Services  " 
for a Windows 10 (or Windows 11) PC