SKAARHOJ/skaarhoj-updater-releases #22: Error each time Skaarhoj updater opens

Mazag, 2022-01-27:
Background Issue: 
      
```
Error: dns service error: service not running
    at new Browser (C:\Users\alainp\AppData\Local\Programs\SKAARHOJUpdater\resources\app.asar\node_modules\mdns\lib\browser.js:88:10)
    at Object.create [as createBrowser] (C:\Users\alainp\AppData\Local\Programs\SKAARHOJUpdater\resources\app.asar\node_modules\mdns\lib\browser.js:116:10)
    at Timeout.ce [as _onTimeout] (C:\Users\alainp\AppData\Local\Programs\SKAARHOJUpdater\resources\app.asar\background.js:2:6933573)
    at listOnTimeout (internal/timers.js:531:17)
    at processTimers (internal/timers.js:475:7)
```

---

SKAARHOJUpdater 1.0.11
Electron 6.1.10
win32 10.0.19043
Locale: fr

Mazag, 2022-01-27:
Hi,

I tried to configure my ATEM TCP Link, but when I start the Firmware updater it shows this message, and I can't use any Online Configuration options in the dash board.

What's wrong?

Thanks for your support.

Kind regards,

Alain

s00500, 2022-01-27:
Hi @Mazag 

The error comes from missing bonjour services for the new discovery section. The message will be improved in the next version.

For now please install the package by apple mentioned here: https://github.com/SKAARHOJ/skaarhoj-updater-releases/issues/20

For using online config with the ATEM TCP Link: This product does not support Online Config, as it is not a Unisketch device. Please contact support@skaarhoj.com for more information.

Greetings

closed 2022-01-27 by s00500