SKAARHOJ/skaarhoj-updater-releases #25: Error: dns service error: service not running

gubrist, 2022-01-29:
Background Issue: 
      
```
Error: dns service error: service not running
    at new Browser (C:\Users\Carlos\AppData\Local\Programs\SKAARHOJUpdater\resources\app.asar\node_modules\mdns\lib\browser.js:88:10)
    at Object.create [as createBrowser] (C:\Users\Carlos\AppData\Local\Programs\SKAARHOJUpdater\resources\app.asar\node_modules\mdns\lib\browser.js:116:10)
    at Timeout.ce [as _onTimeout] (C:\Users\Carlos\AppData\Local\Programs\SKAARHOJUpdater\resources\app.asar\background.js:2:6933573)
    at listOnTimeout (internal/timers.js:531:17)
    at processTimers (internal/timers.js:475:7)
```

---

SKAARHOJUpdater 1.0.11
Electron 6.1.10
win32 10.0.19044
Locale: de

s00500, 2022-01-29:
Please check: https://github.com/SKAARHOJ/skaarhoj-updater-releases/issues/20

closed 2022-01-29 by s00500

gubrist, 2022-01-29:
You can either savely ignore it
---> no, I cannot connect my Skaarhoj PTZ Fly any more (Windows 10)
---> The same problem happens on the brand new PC of my wife (Windows 11)

Install this package from here: https://support.apple.com/kb/dl999?locale=en_US
---> I am puzzled to have to install a software from Apple " Download Bonjour Print Services "
for a Windows 10 (or Windows 11) PC

s00500, 2022-01-29:
Ok some options: Please read the full headline of the link:
"Download Bonjour Print Services **for Windows** v2.0.2"

Please install the package, this is needed for the new discovery section, it is an mdns driver that Apple provides for windows users, as windows does not ship with this.

I know that the error is missleading and it is a bummer that it does not work without this, it will be improved in a future update that we are working on

Second option: download an older version