SKAARHOJ/skaarhoj-updater-releases #27: BUG SKAARHOJ UPDATER

Hugoq23, 2022-02-03:
Background Issue: 
      
```
Error: dns service error: service not running
    at new Browser (C:\Users\PC\AppData\Local\Programs\SKAARHOJUpdater\resources\app.asar\node_modules\mdns\lib\browser.js:88:10)
    at Object.create [as createBrowser] (C:\Users\PC\AppData\Local\Programs\SKAARHOJUpdater\resources\app.asar\node_modules\mdns\lib\browser.js:116:10)
    at Timeout.ce [as _onTimeout] (C:\Users\PC\AppData\Local\Programs\SKAARHOJUpdater\resources\app.asar\background.js:2:6933573)
    at listOnTimeout (internal/timers.js:531:17)
    at processTimers (internal/timers.js:475:7)
```

---

SKAARHOJUpdater 1.0.11
Electron 6.1.10
win32 10.0.19042
Locale: fr

s00500, 2022-02-04:
Hi Hugo, please check this issue and install the necessary driver for discovery, we are working on an update that will create a better error message

https://github.com/SKAARHOJ/skaarhoj-updater-releases/issues/20

closed 2022-02-04 by s00500

Hugoq23, 2022-02-04:
All clear thanks
 <http://eyeshot.fr/>	Hugo Queste
Technical Director
Directeur Technique
  ***@***.*** ***@***.***>
 +33 1 49 24 99 78 <applewebdata://E44496F9-13FB-4D69-BDF1-39281FDCB974#>             
 +33 6 34 23 74 70 <applewebdata://E44496F9-13FB-4D69-BDF1-39281FDCB974#>             
 <http://www.eyeshot-av.com/>             
 <https://www.facebook.com/Eyeshot-128127063870781/>             
 <https://www.instagram.com/eyeshot_av>