SKAARHOJ/skaarhoj-updater-releases #28: Update issue

SedonaUAS, 2022-04-19:
Background Issue: 
      
```
Error: destinationFile or packageFile contains illegal chars
    at t.newError (C:\Users\Jayson's MSI Laptop\AppData\Local\Programs\SKAARHOJUpdater\resources\app.asar\background.js:1:11056)
    at Object.task (C:\Users\Jayson's MSI Laptop\AppData\Local\Programs\SKAARHOJUpdater\resources\app.asar\background.js:1:176424)
    at m.executeDownload (C:\Users\Jayson's MSI Laptop\AppData\Local\Programs\SKAARHOJUpdater\resources\app.asar\background.js:1:93014)
```

---

SKAARHOJUpdater 1.0.5
Electron 6.1.10
win32 10.0.19043
Locale: en-US

s00500, 2022-04-19:
Hi @SedonaUAS, can you give some explanation what happened when you got to this error ?

Greetings

SedonaUAS, 2022-04-20:
I had so many errors yesterday I don’t know which on you are referring to. 

 

The most important issue I need to solve is this one that I emailed to your office yesterday. 

 

 

PTZ Fly - Atem - Tally lights - JVC ptz

 

 

Hello, 

I keep having this problem where my Skaarhoj PTZ Fly looses it ability to display the Red and Green tally information. Today I was able to try 10 different things to see what the issue is. 

 

The on camera tally lights on the JVC PTZ are also effected but this issue.

 

Things I have tried that didn’t work:

 

Updating to 1.0.12           

Changing to an early configuration and changing back.

Tried plugging the USB into a different computer and that temporarily worked 

Then I updated this computer to 1.0.12 and it stopped working again

Down graded to 1.0.11 then 1.0.10 and it still didn’t work

I tried deleting and rebuilding the tally lines in the online configuration

It was only when I saved in the Local configuration did the tally start working again. 

I tried several more times to power down the entire system to prove that it would boot up with the tallies working. But they didn’t

Only when I connected the USB and opened the skaarhoj updater could I get the tallies to work again

 

I need the ptx fly to boot up with just the POE network connection and have the tally lights work

                

I created a 90 second video showing this issue

https://youtu.be/6-JVXkqKG0w

 

What am I doing wrong?

 

Jayson

 

 

   928 239 2300

   Sedona, AZ 86351

 ***@***.***> ***@***.***

 <http://www.sedonauas.com/> www.SedonaUAS.com

s00500, 2022-11-21:
Hi @SedonaUAS did this ever work out for you ? Otherwise please address support@skaarhoj.com as the issue seems to be more than just the updater, also try with the latest release again

closed 2022-11-21 by s00500

SedonaUAS, 2022-11-21:
Thank you for the follow up email, because I have not fixed the issue but I have simply been doing a work around each time I use the system. 

I don't have access to the full system except for one day before each show, so it is very difficult to troubleshoot. The next time I will have access will be December 6th. 

I will have plenty of free time to work on this issue at that time. 

⁣Jayson

928-239-2300
***@***.***
www.SedonaUAS.com​

***@***.*** as the issue seems to be more than just the
***@***.***>