SKAARHOJ/skaarhoj-updater-releases #30: AppImage does not launch on Ubuntu 22.04

yannok, 2022-10-25:
When launching the AppImage on Ubuntu 22.04 (all packages are updated), there is an related to the NodeJS version : 
```
$ ./SKAARHOJUpdater.AppImage 
A JavaScript error occurred in the main process
Uncaught Exception:
Error: The module '/tmp/.org.chromium.Chromium.WVLEn2'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 93. This version of Node.js requires
NODE_MODULE_VERSION 99. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`).
    at process.func [as dlopen] (node:electron/js2c/asar_bundle:5:1800)
    at Object.Module._extensions..node (node:internal/modules/cjs/loader:1199:18)
    at Object.func [as .node] (node:electron/js2c/asar_bundle:5:2027)
    at Module.load (node:internal/modules/cjs/loader:988:32)
    at Module._load (node:internal/modules/cjs/loader:829:12)
    at Function.c._load (node:electron/js2c/asar_bundle:5:13331)
    at Module.require (node:internal/modules/cjs/loader:1012:19)
    at require (node:internal/modules/cjs/helpers:94:18)
    at Object.<anonymous> (/tmp/.mount_SKAARHmNDIp8/resources/app.asar/node_modules/bossajs/index.js:4:15)
    at Module._compile (node:internal/modules/cjs/loader:1116:14)
[14082:1025/153752.688925:ERROR:sandbox_linux.cc(376)] InitializeSandbox() called with multiple threads in process gpu-process.

```

s00500, 2022-10-31:
Hi @yannok, thanks for the report, we will check what went wrong...

s00500, 2022-10-31:
Could you try this file ? https://we.tl/t-TzSsoJB7J5

yannok, 2022-11-10:
Hi @s00500 , sorry I was AFK for one week, the WeTransfer file has expired. Could you re-post it? Thanks

yannok, 2022-11-10:
Thanks, it works now. I cannot currently connect the devices but at least the app starts without issue.

Note that I initially got the following error : 
`Error: libdns_sd.so.1: cannot open shared object file: No such file or directory`

that I solved by installing the missing lib:
`sudo apt install libavahi-compat-libdnssd1`

If I understand well the concept of AppImage, it should contain all its dependencies so this step should not be necessary.

s00500, 2022-11-15:
We will not ship the libavahi stuff with the package, it is also only needed for the discovery section on the main page...

I will update the latest release with the new build, thanks for the report and test :-)

closed 2022-11-15 by s00500

yannok, 2022-11-17:
just my two cents about this : even if the libavahi-compat-libdnssd1 is only needed for the discovery section, the app will not start at all if the lib is not installed on the system so you should either add it as a requirement or include it in the AppImage (the latter solution would be closer to the concept of AppImage that should not depend on the packages installed on the system)

s00500, 2022-11-21:
I agree but unfortunately my resources are limited for working on this