SKAARHOJ/skaarhoj-updater-releases #31: Launch Error on Ubuntu 22.04.1

yannok, 2022-11-18:
Background Issue: 
      
```
Error: getaddrinfo -3001
    at errnoException (/tmp/.mount_SKAARHZ5f8oe/resources/app.asar/node_modules/mdns/lib/resolver_sequence_tasks.js:199:11)
    at getaddrinfo_complete (/tmp/.mount_SKAARHZ5f8oe/resources/app.asar/node_modules/mdns/lib/resolver_sequence_tasks.js:112:10)
    at GetAddrInfoReqWrap.oncomplete (/tmp/.mount_SKAARHZ5f8oe/resources/app.asar/node_modules/mdns/lib/resolver_sequence_tasks.js:120:9)
```

---

SKAARHOJUpdater 1.0.14
Electron 16.0.7
linux 5.15.0-53-generic
Locale: en-US

s00500, 2022-11-21:
does this still happen with the installed package you mentioned in #30 ?

yannok, 2022-11-29:
yes, the problem occurs with the latest release mentioned in #30? The app starts but nothing works at all.

s00500, 2024-04-28:
hey @yannok, did you try this again at any time ?