SKAARHOJ/skaarhoj-updater-releases #32: First install - DNS error on startup

ofilms-gareth, 2023-03-10:
Background Issue: 
      
```
Error: dns service error: service not running
    at Array.DNSServiceGetAddrInfo (C:\Users\gareth\AppData\Local\Programs\SKAARHOJUpdater\resources\app.asar\node_modules\mdns\lib\resolver_sequence_tasks.js:92:14)
    at next (C:\Users\gareth\AppData\Local\Programs\SKAARHOJUpdater\resources\app.asar\node_modules\mdns\lib\browser.js:109:21)
    at MDNSService.on_resolver_done (C:\Users\gareth\AppData\Local\Programs\SKAARHOJUpdater\resources\app.asar\node_modules\mdns\lib\resolver_sequence_tasks.js:33:11)
    at SocketWatcher.MDNSService.self.watcher.callback (C:\Users\gareth\AppData\Local\Programs\SKAARHOJUpdater\resources\app.asar\node_modules\mdns\lib\mdns_service.js:18:40)
```

---

SKAARHOJUpdater 1.0.14
Electron 16.0.7
win32 10.0.19045
Locale: en-US

s00500, 2023-03-10:
Hello @ofilms-gareth,

Please refer to the explanation and solution from this ticket

-> https://github.com/SKAARHOJ/skaarhoj-updater-releases/issues/20

Greetings,
Lukas

closed 2023-03-10 by s00500

ofilms-gareth, 2023-03-10:
Thanks Lukas.  I've installed Bonjour and will test when I get a chance.  For your info, I wasn't able to detect the RCPv2 at all without Bonjour.