SKAARHOJ/skaarhoj-updater-releases #35: not sure what to do

SH010101, 2023-08-09:
Background Issue: 
      
```
Error: dns service error: bad reference
    at SocketWatcher.MDNSService.self.watcher.callback (C:\Users\steve\AppData\Local\Programs\SKAARHOJUpdater\resources\app.asar\node_modules\mdns\lib\mdns_service.js:18:40)
```

---

SKAARHOJUpdater 1.0.14
Electron 16.0.7
win32 10.0.22621
Locale: en-US

s00500, 2023-08-21:
Hi @SH010101, have you installed the apple bonjour drivers ?

SH010101, 2023-08-21:
I have tried.  The last thing I remember about it was it being a  PRINTING component.   Maybe I have the wrong driver.    I honestly haven’t had a chance to work on it again in a while.   I plan to look at it this evening.    Thanks for checking in.   Steven Hollis336.608.8486HollyDogMedia.com

s00500, 2023-08-22:
Ok, so are you trying to get the stuff going, could you use the rest of the updater normally ?

s00500, 2024-04-28:
stale, closing

closed 2024-04-28 by s00500