SKAARHOJ/skaarhoj-updater-releases #36: plz SKAARHOJ

1HollyDog, 2023-09-09:
Background Issue: 
      
```
Error: write EIO
    at afterWriteDispatched (node:internal/stream_base_commons:164:15)
    at writeGeneric (node:internal/stream_base_commons:155:3)
    at WriteStream.Socket._writeGeneric (node:net:780:11)
    at WriteStream.Socket._write (node:net:792:8)
    at writeOrBuffer (node:internal/streams/writable:389:12)
    at _write (node:internal/streams/writable:330:10)
    at WriteStream.Writable.write (node:internal/streams/writable:334:10)
    at console.value (node:internal/console/constructor:289:16)
    at console.warn (node:internal/console/constructor:368:26)
    at consoleLog (/Applications/SKAARHOJUpdater.app/Contents/Resources/app.asar/node_modules/electron-log/src/transports/console.js:100:17)
```

---

SKAARHOJUpdater 1.0.14
Electron 16.0.7
darwin 22.3.0
Locale: en-US

s00500, 2023-09-10:
Hello @1HollyDog 
Seems like there was an issue writing the log file, could you explain a bit more what happened ?

1HollyDog, 2023-09-10:
Thanks for the follow up.  I THINK I have it working now.   This thing has challenged me at every level.   🤓

s00500, 2023-09-11:
Ok

closed 2023-09-11 by s00500