SKAARHOJ/skaarhoj-updater-releases #39: cant connect to bluepill server on mac osx, chrome

yaniv054, 2024-09-22:
Background Issue: 
      
```
Error: Invalid URL
    at IpcMainImpl.<anonymous> (/Applications/SKAARHOJUpdater.app/Contents/Resources/app.asar/background.js:2:566053)
    at IpcMainImpl.emit (node:events:394:28)
    at Object.<anonymous> (node:electron/js2c/browser_init:161:9959)
    at Object.emit (node:events:394:28)
```

---

SKAARHOJUpdater 1.0.14
Electron 16.0.7
darwin 20.6.0
Locale: en-US

yaniv054, 2024-09-25:
> Hi @yaniv054 <https://github.com/yaniv054>,
> Could you please provide a little more detail to your issue ? The firmware
> updater should not be needed to connect to your bluepill server. Ideally
> you should simply open the IP in a browser

I connect the blue pill server to an unmanaged POE switch. The controller,
computer and the BMD switch are also connected.
when I type the IP from the blue pill to a  browser, I get the timeout
message.

when I connect the blue pill to a different network, everything works fine
and I have access.
I tried fixed IP and DHCP none of it works on the first network.
both networks are super simple with basic topology

s00500, 2024-09-25:
First of all I have seen such issues before with IP conflicts...

Second I would kindly ask to send this request to support@skaarhoj.com as it is unrelated to the firmware updater :-)

Thanks